mgerhard74/amis_smartmeter_reader #4: Verbrauch via MQTT

csc-solution, 2024-06-02:
Gibt es eine Möglichkeit, den Tagesverbrauch bzw Tagesbezug / Tageslieferung per MQTT ausgeben zu lassen? Das würde die Anzeige eines Graphen in HomeAsistant erleichtern.

Danke im Voraus
Tom

tiotuba, 2024-06-03:
hi @csc-solution soweit ich weiß, wenn man auf der Web-Interface den MQTT Broker einstellt kann man das in HomeAssistant empfangen und in das Energy Dashboard einbinden.

Home Assistant Entities
[image]

Home Assistant Energy Dashboard
[image]

[image]

AMIS MQTT Konfiguration:
[image]

csc-solution, 2024-06-03:
Hi,

danke für die Info. Habe das Energy Dashboard nicht in Verwendung und habe nicht daran gedacht :) Werde es einmal damit versuchen.

Grüße
Tom