mgerhard74/amis_smartmeter_reader #5: Mehrere Menüpunkte nach Update auf 1.2.8 nicht mehr funktionsfähig

k3067e3, 2024-07-16:
Habe den AMIS Reader schon mehrere Jahre und auch immer wieder Updates gemacht.
Nach dem Update von 1.2.7 auf 1.2.8 funktionieren nur noch die Menüpunke:
Zähler, Grafik, Status, Allgemein, wobei bei Allgemein der untere Bereich so aussieht

[image]

Unter Log, WIFI, MQTT, Update, Versionsinfo bleiben die Seiten leer.
Mehrfacher Neustart erfolglos, unterschiedliche Browser probiert, STRG+F5 für erzwungenen Refresh bringt auch nichts.

Kann man hier über die Konsole vom Browser mit einem speziellen Befehl, so wie er für den Test der IR Diode erklärt ist, wieder was reparieren?

mgerhard74, 2024-07-16:
Hallo,
kannst du die Dateien aus http://www.mitterbaur.at/amis/firmware1-2-8.zip (bis auf firmware.bin und littlefs.bin) einzeln neu hochladen und neu starten?
Falls die Probleme weiterhin bestehen: Wenn du keinen USB/TTL Programmer zum Flashen hast, kannst gerne vorbeikommen oder mir den AMIS Reader zum Flashen zusenden. Adresse lt http://www.mitterbaur.at/impressum.html 
lg Gerhard

k3067e3, 2024-07-16:
Leider bringt klicken auf Update im Menü nur eine leere Seite, somit geht nichts über das GUI für neuerlichen Upload.

Meinst du ich soll die einzelnen Dateien über einen USB/TTL Programmer flashen?
So ein Teil müsst ich wo haben, das kann ich gerne probieren.

mgerhard74, 2024-07-16:
Hier die Anleitung zum Flashen: 
http://www.mitterbaur.at/amis/AmisReaderflashen.pdf

lg Gerhard

> Leider bringt klicken auf Update im Menü nur eine leere Seite, somit 
> geht nichts über das GUI für neuerlichen Upload.
> 
> Meinst du ich soll die einzelnen Dateien über einen USB/TTL Programmer 
> flashen?
> So ein Teil müsst ich wo haben, das kann ich gerne probieren.
> 
> [2].

Links:
------
[1] 
https://github.com/mgerhard74/amis_smartmeter_reader/issues/5#issuecomment-2230070072
[2] 
https://github.com/
--=_259fb2a23e1c31a149c8f86cf96ff6b7
Content-Type: multipart/related;
 boundary="=_0056fc27ba513df1a6fab0454ccddb42"

--=_0056fc27ba513df1a6fab0454ccddb42
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hier die Anleitung zum Flashen: <a href=3D"http://www.mitterbaur.at/amis=
/AmisReaderflashen.pdf">http://www.mitterbaur.at/amis/AmisReaderflashen.pdf=
</a></p>
<p>lg Gerhard</p>
<p id=3D"reply-intro">Am 16.7.2024 07:43, schrieb k3067e3:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<p dir=3D"auto">Leider bringt klicken auf Update im Men&uuml; nur eine leer=
e Seite, somit geht nichts &uuml;ber das GUI f&uuml;r neuerlichen Upload.</=
p>
<p dir=3D"auto">Meinst du ich soll die einzelnen Dateien &uuml;ber einen US=
B/TTL Programmer flashen?<br />So ein Teil m&uuml;sst ich wo haben, das kan=
n ich gerne probieren.</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
gerhard74/amis_smartmeter_reader/issues/5#issuecomment-2230070072" target=
=3D"_blank" rel=3D"noopener noreferrer">view it on GitHub</a>, or <a href=
2DZMSXH3AVCNFSM6AAAAABK5YCGDOVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDEMZQ=
GA3TAMBXGI" target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<=
le=3D"color: transparent; font-size: 0; display: none; visibility: hidden; =
overflow: hidden; opacity: 0; width: 0px; height: 0px; max-width: 0; max-he=
eader/issues/5/2230070072</span><span>@</span><span>github</span><span>.</s=
pan><span>com&gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_0056fc27ba513df1a6fab0454ccddb42
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_0056fc27ba513df1a6fab0454ccddb42--

--=_259fb2a23e1c31a149c8f86cf96ff6b7--

k3067e3, 2024-07-16:
Nach FW Update über USB/TTL Programmer funktioniert wieder alles, danke für die Unterstützung!

closed 2024-07-16 by k3067e3