mgerhard74/amis_smartmeter_reader #6: MQTT connection is not working

superwolfi82, 2024-08-13:
Hallo Zusammen

Ich habe zwei AMIS Reader mit der Version 1.2.7 und habe das problem, dass via MQTT keine Daten an Homeassistant (mosquitto) gesendet werden, die Abfrage von homeassistant via Rest funktioniert ohne probleme.

vom Amis-reader scheint die connection  successful established
[image]

auch auf Mesquite seite im debug sehe ich die connection "MQTT ping" messages  aber leider keine publish messages

[image]

2024-08-13 12:45:32: Received PINGREQ from Amis-normalstrom
2024-08-13 12:45:32: Sending PINGRESP to Amis-normalstrom
2024-08-13 12:45:35: Received PINGREQ from Amis-warm
2024-08-13 12:45:35: Sending PINGRESP to Amis-warm

superwolfi82, 2024-08-13:
kleines update -- ich lasse das ganze drinnen, es scheint so als braucht der mqtt stream ca 15 min is daten gesendet werden.. 

[image]

maxus-git, 2024-08-13:
Hallo superwolfi82,

dreh deinen "MQTT Keep Alive" von 900 mal auf 1 Sekunde ;-)
(900 / 60 = 15min)

LG Markus

mgerhard74, 2024-08-13:
Danke Markus, 
wollte ich auch gerade antworten. lg Gerhard

superwolfi82, 2024-08-13:
@maxus-git  @mgerhard74   Danke .. :-) 

ist ein bisschen verwirrend wenn das MQTT Keep Alive als sende intervall doppelt verwendet wird  (ja es steht im kleingedruckten...)

gibt es eine Möglichkeit bei jedem signal vom stromzähler dies übertragen zu bekommen  ?

mgerhard74, 2024-08-13:
Ja, stelle 1 sec. ein.
lg Gerhard

superwolfi82, 2024-08-13:
danke!

maxus-git, 2024-08-13:
@mgerhard74
Wäre eine Idee für ein künftiges Update, das Publish-Intervall vom Keep Alive zu trennen.

mgerhard74, 2024-08-13:
Webinterface angepasst: Feld auf "MQTT Publish-Intervall/Keep Alive" geändert.

closed 2024-08-13 by mgerhard74