mgerhard74/amis_smartmeter_reader #7: Fronius SmartMeter Ersatz - gerechnete AC Ströme für Load Balancing mit WattPilot

grisu122, 2024-08-17:
Hallo Zusammen,

Ich habe einen Vorschlag für einen (möglichen) zusätzlichen Use Case:

Der Fronius WattPilot dürfte die AC-Ströme die vom SmartMeter übermittelt werden für das Load Balancing verwenden (Konfiguration im WattPilot erfolgt in Ampere). Bei einem Versuch wurden zwar die Solar- & Netzleistung in der App korrekt angezeigt, es erfolgte jedoch keine Anpassung der Ladeleistung, da der Phasenstrom (oder Phasenleistung) nicht zur Verfügung steht (bzw. als 0 A / 0 W übertragen wird).

Für das Load Balancing mit dem WattPilot könnten die gerechneten Phasenströme vermutlich ausreichen.  
Ix = (1.7.0 - 2.7.0) / 400 / Wurzel(3)  
oder näherungsweise: Ix = power / 692.82

Bitte um Prüfung ob die 3 Phasenströme implementiert werden können. Danke

mgerhard74, 2024-08-18:
Hallo @grisu122 
das scheint machbar. Aktuell wenig Zeit. Das Dokument könnte hilfreich sein:

[Smart_Meter_Register_Map_Float.xlsx](https://github.com/user-attachments/files/16647659/Smart_Meter_Register_Map_Float.xlsx)

lg Gerhard

mgerhard74, 2024-08-19:
Hallo @grisu122 
ich glaub ich habs korrekt umgesetzt. Lt RMMS schauts gut aus. Könnte ich dir die Firmware.bin per Mail zum Testen schicken?
lg Gerhard

grisu122, 2024-08-24:
Hallo Gerhard,
danke fürs erstellen der Test Firmware. Es dürfte dann scheinbar doch nicht so einfach sein. Mit v1.2.9 wird die Leistung stabil von / an den Wechselrichter übertragen. Mit der Test-firmware wechselt der Status des SmartMeter beim Wechselrichter im Rythmus von 5 bis 10 Sekunden zwischen angezeigter SmartMeter-Leistung und der Meldung "Der Zähler wurde erkannt, liefert jedoch keine Werte. Prüfen Sie die Verkabelung". 
Der WattPilot selber startet auch, schaltet dann aber wieder ab, wenn der Zähler (SmartMeter) verlorgen geht. Die Laufzeit ist zu kurz um prüfen zu können ob eine Regelung erfolgt. Auf alle Fälle mal Danke fürs erstellen der Test Firmware.
Ich werde mal versuchen herauszufinden was diese Fehlermeldung am Wechselrichter verursacht. Eventuell erfolgen da Plausibilitätsprüfungen der Werte U, I, P und S am Wechselrichter.
lg grisu122

mgerhard74, 2024-08-25:
[meter_cpp.txt](https://github.com/user-attachments/files/16739264/meter_cpp.txt)
Hallo @grisu122 ,
bei mir gleiches Verhalten - anbei die geänderte meter.cpp. Zeile 153-205 sind neu hinzugekommen. Vielleicht findest du das Problem?
lg Gerhard

mgerhard74, 2024-08-25:
Hallo @grisu122 ,
Ich habe nun den Gesamtstrom nicht übertragen. Mein Fronius bringt die Meldung "Der Zähler wurde erkannt, liefert jedoch keine Werte. Prüfen Sie die Verkabelung" nicht mehr. 
Firmware.bin und meter.cpp anbei.
Bitte um Test.
lg Gerhard

[firmware.zip](https://github.com/user-attachments/files/16740309/firmware.zip)
[meter_cpp.txt](https://github.com/user-attachments/files/16740310/meter_cpp.txt)

mgerhard74, 2024-10-13:
Hallo @grisu122 ,
bitte die Firmware.bin anbei nochmals testen.
Danke.
[firmware.zip](https://github.com/user-attachments/files/17354816/firmware.zip)

mgerhard74, 2026-09-12:
Ich setze diesen Issue nun auf gelöst:
Der Fronius Gen24 braucht nun für das Einspeiselimit auch Einzelspannungen und Ströme und dies funktioniert mit der aktuellen Lesekopf Firmware.

closed 2026-09-12 by mgerhard74