mgerhard74/amis_smartmeter_reader #8: Dupliziertes signal könnte stören?

GameTec-live, 2024-09-14:
Ich versuche verzweifelt die messdaten irgendwie aus meinen Zähler zu bekommen und hab dieses Projekt gefunden;
Aber ich bekomme den Fehler, dass der AMIS Schlüssel invalide ist?
Mein optischer lese/schreibe Kopf sendet auf grund von reflektionen (usw) alles gesendete wieder zurück an den MCU, könnte das vlt ein Störgrung sein? Ist ein deduplikator schon eingebaut? gibt es bekannte tricks zum lösen dieses problems?

mgerhard74, 2024-09-14:
Hallo,
stimmt der MBUS Key mit dem aus dem NetzOÖ Portal überein? Ich habe zu Testzwecken mit Alufolie / Spiegeln ect versucht das ausgesendete IR-Signal wieder auf den Empfänger zu bekommen - es war fast nicht möglich, ich habe mir eine IR - Testschaltung mit IR-Sender und Empfänger zum Testen der AMIS Platinen gebaut. Ev. hast du falsche/nicht kompatible IR-Bauteile verwendet.
Du kannst mir eine eMail schreiben, ich sehe nach, ob ich noch einen Lesekopf für dich über habe.....
lg Gerhard

GameTec-live, 2024-09-14:
Ich verwende den "piggymeter" lesekopf, nur leider unterstützt die software die (deppate) implementierung der NetzOÖ... Der Schlüssel sollte übereinstimmen, da ich den immer copy pasted habe... (Das Duplikations problem habe ich entdeckt weil ich mich einfach mit einem USB-TTL adapter verbunden habe)

Wenn du noch ein funktionierenden lesekopf über hast und den mir geben könntest, wäre das sehr nett. Nur mein Zählerschrank ist ein sehr guter WLAN blocker (sobald ich die Tür zuklappe is das signal weg, mein plan war einfach das UART signal durch den spalt zu legen und dann den ESP außen an die wand (wo auch eine steckdose ist) zu packen) und da deiner den ESP fest einlötet wäre das ein problem?

mgerhard74, 2024-09-15:
Hallo,
wegen dem Lesekopf habe ich dir eine EMail geschrieben. Ich würde einen Wifi Range Extender in der Nähe vom Zählerkasten empfehlen oder eine externe Antenne am ESP anlöten (Elektronikkentnisse notwendig).
lg Gerhard

closed 2024-09-15 by mgerhard74