mgerhard74/amis_smartmeter_reader #10: Wifi Steckdose schalten

Wahnsinicka, 2025-02-18:
Hallo Gerhard!

Ich hätte gerne die Wifi Steckdosen  - Funktion getestet. Dazu hab ich meine AMIS-Reader von 1.2.7 auf 1.3.1 (bzw 1.3.0) aktualisiert.

Leider habe ich die Funktion nicht gefunden. Im Sourcecode (Index.html Zeile 321-342) hab ich dann gesehen das dies unter "Allgemein" angezeigt werden sollte.

Muss dazu eine Funktion aktiviert/deaktiviert werden?

Vielen Dank!

[image]

Wahnsinicka, 2025-02-18:
Ps. auch die Versionsinfo zeigt nicht den aktuellen stand: (getestet mit Inkognito Browserfenster)

[image]

[image]

Wahnsinicka, 2025-02-18:
Hatte eine alte Index.html im Upload.
Aus [Source code.zip](https://github.com/mgerhard74/amis_smartmeter_reader/archive/refs/tags/v1.3.1.zip) alle Dateien aus \data einzeln hochladen und abschließend den AMIS Reader noch einmal neu starten.

mgerhard74, 2025-02-18:
Hallo, 
hast du die Funktion nun nach einem Upload der korrekten index.html Datei?
Meine Möglichkeiten hier im Urlaub sind leider eingeschränkt.
Lg Gerhard

Wahnsinicka, 2025-02-18:
Funktionen sind alle hier. Issue kann geschlossen werden. Danke&schönen Urlaub!

closed 2025-02-18 by Wahnsinicka