mgerhard74/amis_smartmeter_reader #13: Wochendaten fehlen

meks007, 2025-03-03:
Tut mir leid, wenn ich hier nochmal lästig sein muss.

Ich hab gestern Nachmittag auf 1.4.1 aktualisiert.
[image]

[image]

Ebenfalls gelöscht habe ich den Browser Cache mit STRG+F5

Ich hab noch immer mein Problem mit den Daten Energie wöchentlich.
Hab die Wochendaten über die Schaltfläche bei den Logs gelöscht, hatte gestern nur die Anzeige unter heute, mit meinen aktuellen Zählerdaten.

Hab gedacht ich warte auf heute, aber hat nichts geändert.

Irgend eine Idee wie sich die Wochendaten wieder befüllen?

[image]

Noch ein weiteres Details am Rande, habe weil ich es probieren wollten den Reboot um Mitternacht aktiviert. 

[image]

Danach auch nochmal neu gestartet, gestern gegen 21h.

Der Reboot um Mitternacht ist aber nicht passiert...

_Originally posted by @k3067e3 in https://github.com/mgerhard74/amis_smartmeter_reader/issues/12#issuecomment-2693608981_

k3067e3, 2025-03-04:
Ich hab mal, in der Hoffnung dir Arbeit zu ersparen, sicherheitshalber den Reader mit dem USB/TTL Programmer komplett neu geflasht.

Jetzt habe ich für den heutigen Tag einen plausiblen Eintrag bei den Wochendaten.

[image]

Ich habe auch alle Konfig Daten nochmal schnell manuell eingegeben, einzig und allein meine Monatsdaten habe ich wieder über das Menü restored.

Ich bin zuversichtlich, dass sich morgen dann auch die Wochendaten weiter sinnvoll befüllen, wenn schon beim heutigen Tag was passendes drinnen ist.

Ich werde dazu berichten, ebenso ob der 0h Reboot greift.

mgerhard74, 2025-03-04:
Der 0Uhr Reboot rebootet nur, wenn der Reader länger als 12h läuft. Lg Gerhard

mgerhard74, 2025-03-04:
Hallo zusammen,
ich habe eben meinen Reader zurück auf 1.3.1 per index.html, cust.js und firmware.bin Update + Neustart geflasht. 
Dann wieder auf 1.4.1 mit Firmware.bin + Neustart, dann index.html und cust.js Update + Neustart. 
Wochendaten sind korrekt befüllt und die Lieferung bei "heute" zählt weiter hoch. 
Somit kann ich bei meinem Setup kein Problem nachstellen.
lg Gerhard

rorso, 2025-03-05:
Nur zur Gegenprobe - ich habe irgendwann im Februar wieder mal ein Update gemacht. Zu der Zeit auf 1.3.1. Mit dem Image sehen alle Statistiken eigentlich vernünftig aus. Dass die durchschnittliche Leistung leer ist, war mir allerdings nicht aufgefallen. Ich habe keine Ahnung, ob das vorher richtig war.

Da ich meine Auswertungen ohnehin über MQTT und HomeAssistant mache, hatte ich dafür bisher auch keine Verwendung.

Ich habe nur das `firmware.bin` und anschließend die `cust.js` und `index.html` hochgeladen.

[image]

[image]

mgerhard74, 2025-03-05:
Hallo @meks007 ,
wo kann ich denn ansetzen, um die kleinen Probleme zu korrigieren? Ich bin leider in der ESP Programmierung nur Anfänger.
Aktuell ist folgendes bekannt:
- Zählerseite refresht manchmal nach zig Sekunden die Wirkleistungswerte nicht mehr
- Teilweise Probleme beim Thingspeak Upload
- Teilweise Wochendatenanzeige fehlerhaft
lg Gerhard

meks007, 2025-03-05:
Ich bin mir relativ sicher, dass das nicht an den Änderungen von mir liegt. Da ich zu 1.3.1 keine verlässlichen Sources habe, kann ich ehrlicherweise keine Auskunft geben. Meine Vermutung ist, dass neben den Javascript-Files auch die Firmware-Files in Github nicht aktuell. Das heißt ich benötige zumindest verlässliche 1.3.1 Sources um gegen 1.4 zu comparen und Differenzen herauszufinden. Auf Github kann ich mich nicht verlassen, weil wir da ja schon Differenzen zu deiner Homepage festgestellt haben.

mgerhard74, 2025-03-05:
Sorry, ich hätte schon versucht den Quellcode 1.3.1 lokal und hier gleich zu halten. 
Ich habe meinen Quellcode, mit welchem ich V1.3.1 erstellt habe, hier online gestellt: http://mitterbaur.at/amis/Sourcecode1-3-1.zip
lg Gerhard

meks007, 2025-03-05:
Es gibt keine Differenzen in den Sourcen zwischen 1.3.1 und 1.4. Daran liegt es also nicht. Meine Vermutung ist aktuell, dass es irgendwo noch einen Unterschied in den verwendeten Libraries gibt, weil ich ja in #11 das Problem hatte, das die verwendeten Libraries in der platformio.ini nicht eindeutig waren und Interpretationen zugelassen haben. Ich habe jetzt versucht über den 1.3.1 Source "korrektere" Bibliotheken abzuleiten und habe mal ein firmware.bin erstellt und bei mir geflasht.

Ich würde die Diskussion dazu gerne im PR #15 weiterführen.

mgerhard74, 2025-03-05:
Danke, gerne

closed 2025-03-05 by mgerhard74