mgerhard74/amis_smartmeter_reader #14: Thingspeak mit v1.4.1- kein Update zu Thingspeak

mgerhard74, 2025-03-04:
Ein Bekannter hat auch schon auf 1.4.1 aktualisiert. Danach funktionierte der Thingspeak Upload nicht mehr. 
Bei mir allerdings schon.

k3067e3, 2025-03-04:
Ich hab ja heute früh meinen Reader mit dem USB TTL Programmer komplett neu geflasht, Konfig manuell eingeben, aber die Config.amis vorher vs nachher mit Notepad ++ verglichen, um Tippfehler auszuschließen. Bei mir sagt die Thingspeak Webseite auch letzter Eintrag vor 14h...

k3067e3, 2025-03-05:
Ich hab jetzt nochmal mit USB/TTL geflasht, V 1.3.1, Quelle http://www.mitterbaur.at/amis/firmware1-3-1.zip

[image]

Damit klappt auch der Thingspeak Upload

[image]

k3067e3, 2025-03-07:
Habe jetzt von 1.3.1 auf 1.4.2 aktualisiert via GUI.
firmware.bin aktualisiert, Reboot, index.hmtl (wo noch immer V 1.4.1 in den Versionsinfos steht) und cust.js aktualisiert, Reboot gemacht.

[image]

Nach Reboot waren meine verfügbaren Wochendaten da. Leider kein Screenshot gemacht.

Hab dann zum Testen den Ping Test aktiviert, und rebootet.

[image]

Screenshot von 10h35 36sec

[image]

Wenn ich Neustart bei Ping wieder deaktivere, sieht es so aus

[image]

Kann ich mit ein- und auschalten eindeutig reproduzieren.

Wenn Ping aus ist, geht auch der Thingspeak Upload, nach ca. 30 Sekunden Laufzeit, mit aktivierten Ping auch nach 3 Minuten noch nichts

mgerhard74, 2025-03-07:
@k3067e3 , kannst du bitte das Update v1.4.3 versuchen, bei mir gibts keine Probleme mehr, wenn Ping restart check aktiviert ist.
Von 1.4.2 nur die firmware.bin aktualisieren.
lg Gerhard

k3067e3, 2025-03-07:
Gerade getestet, schaut bei mir jetzt auch gut aus.

LG Markus