mgerhard74/amis_smartmeter_reader #16: Zeitstempel zum JSON hinzufügen

Poldi8488, 2025-04-21:
Wäre es möglich, das JSON (welches per mqtt gesendet wird) um einen unix-timestamp zu ergänzen? 
Da die Uhrzeit (Zähler) auf der Webpage ausgegeben wird, liegt diese doch schon vor...

Hintergrund: Wenn ich die Daten in einer Hausautomatisierung auswerte kann ich einen Plausibilitäts-Check über die Uhrzeit ausführen und ggf. einen Fehlfunktion/Ausfall des AMIS-Leser erkennen.

LG Arnold

Poldi8488, 2025-04-22:
@rorso 
Danke für die Anregung! Ich konnte in IOBroker etwas Ähnliches programmieren.
Also bei jeder Werteaktualierung die akutelle Zeit als Unix-Timestamp in einen Datenpunkt schreiben und dann im Intervall die Differenz zwischen aktueller Unix-Time und dem Wert der Variable auswerten.

LG Arnold

Poldi8488, 2025-04-27:
@Rorso:
Bzgl. Deiner Dropouts: Falls Du die am PCB-Antenne vom ESP8266 verwendest, kann ich Dir nur das Antennenmod, welches Gerhard auf seiner Webpage beschrieben hat empfehlen. Ich hab eine 2,4GHz-Schraubantenne von einem alten WLAN-Router genommen, mir auf Amazon ein vom Anschluß her passendes 1m Antennenkabel (https://www.amazon.de/dp/B07YCC3HFP?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1) besorgt und das eine Ende davon am ESP-PCB angelötet. Das relativ dünne Antennenkabel hab ich dann durch den Falz der Zählerkastentüre nach außen geführt und siehe da, ich kann nun den ESP ohne zusätzlichen WLAN-Repeater in meiner Wohnung ohne Dropouts empfangen.

mgerhard74, 2025-04-27:
Ich scheitere an einer Funktion zur Konvertierung der Datum/Uhrzeit aus dem Smartmeter ins Unix Epoch Format. 
Für die Anzeige im Webinterface erfolgt die Konverierung in cust.js in der Funktion timeDecoder.
Aus dem Smartmeter kommt in buffer[3]-buffer[8] Datum/Uhrzeit in folgendem Format:
[image]

Ich habe in amis.cpp probiert:

```
#include <TimeLib.h>

time_t cp48ToUnix(uint64_t cp48)
{
  uint8_t second = (cp48 >> 0) & 0x3F;        // bits 1-6
  uint8_t minute = (cp48 >> 8) & 0x3F;        // bits 9-14
  uint8_t hour   = (cp48 >> 16) & 0x1F;       // bits 17-21
  uint8_t day    = (cp48 >> 25) & 0x1F;       // bits 25-29
  uint8_t month  = (cp48 >> 33) & 0x0F;       // bits 33-36
  uint16_t year  = (cp48 >> 30) & 0x7F;       // bits 30, then 32-37 to 40

  // CP48-Jahr ist als Offset zu 2000 zu interpretieren:
  year += 2000;

  // Struktur für die Zeitangabe
  tmElements_t tm;
  tm.Second = second;
  tm.Minute = minute;
  tm.Hour   = hour;
  tm.Day    = day;
  tm.Month  = month;
  tm.Year   = CalendarYrToTm(year); // Konvertiert 4-stellige Jahreszahl zu Offset ab 1970

  // Erzeugt Unix-Zeit
  return makeTime(tm);
}
```

Und der Aufruf in void amis_decoder()
`a_result[9] = cp48ToUnix(buffer[4] + buffer[5]*0x100 + buffer[6]*0x10000 + buffer[7]*0x1000000 + buffer[8]*0x100000000);`

Leider stimmt das Ergebnis nicht.
lg Gerhard

rorso, 2025-04-27:
Wow, was für eine seltsame Datenstruktur.
Ich nehme an, das ist der "now" Wert, der im WebService in dieser Struktur abgefragt wird:

```
{
  "now": "0x343bf7364c",
  "1_8_0": 26230440,
  "2_8_0": 10884717,
  "3_8_1": 3513659,
  "4_8_1": 6696495,
  "1_7_0": 497,
  "2_7_0": 0,
  "3_7_0": 0,
  "4_7_0": 184,
  "1_128_0": 0,
  "uptime": 1275298,
  "things_up": ""
}
```

Den habe ich binär zerlegt und entsprechend der Tabelle analysiert.

Demnach müsste die Zuweisung SO funktionieren:

```
  uint8_t second = (cp48 >> 0) & 0x3F;        // bits 1-6
  uint8_t minute = (cp48 >> 8) & 0x3F;        // bits 9-14
  uint8_t hour   = (cp48 >> 16) & 0x1F;       // bits 17-21
  uint8_t day    = (cp48 >> 24) & 0x1F;       // bits 25-29
  uint8_t month  = (cp48 >> 32) & 0x0F;       // bits 33-36
  uint16_t year  = (cp48 >> 29) & 0x0F + ((cp48 >> 36) & 0x0F) * 8 + 2000;       // bits 30, then 32-37 to 40

```
Zum Einen sind die Shift Parameter für Tag und Monat jeweils eins niedriger, zum Anderen ist das Jahr in zwei Teilen gespeichert. Der `>>29` Teil leifert eine "1", der `>>36` Teil eine "24" --> 25. Die 2000 kommen gar nicht vor, das Jahr ist nach der Definition nur zweistellig in den Daten.

Nach der Variante ließ sich der "now" Wert in ein korrektes Datum übersetzen.

[AmisDatum.ods](https://github.com/user-attachments/files/19932006/AmisDatum.ods)

Zwischen drin sind immer wieder Bits gesetzt, die in der Tabelle nicht gelistet werden.

Ich habe eine minimalistische LibreOffice Calc Tabelle dafür gemacht.

rorso, 2025-04-28:
Ein 'Problem' mit der Uhrzeit könnte noch sein, dass es "lokale Zeit" ist. Die Sommerzeit Info wurde hier (noch) nicht ausgewertet. Aktuell vermute ich, dass auf die Art im Frühling ein Loch in der Zählung auftritt, im Herbst eine Stunde doppelt ausgegeben wird.

Um eine konsistent, streng aufsteigende Timestamp-Angabe für die Datenpunkte zu haben, müsste man die Angabe beispielsweise in UT ausgeben, also die jeweilige (temporäre) Zeitzone heraus rechnen.

Das machen aber auch nicht alle Stromanbieter oder Netzbetreiber in ihrer Ausgabe der 15-Minuten Verbrauchswerte Statistik.

Je richtiger man es machen will, desto komplizierter wird es...

mgerhard74, 2025-05-01:
Hallo Zusammen,
ich habe mit eurer Hilfe eine Testversion mit Ausgabe der Sekunden ab 1.1.1970 in REST und MQTT erstellt. Es wird die Zeit aus dem Smartmeter auf Sekunden umgerechnet, Achtung bei Wechsel auf Sommer- bzw Normalzeit.

@rorso : Danke für deine Hilfe, in dieser Zeile habe ich ein 0x0F durch 0x07 ersetzt, da nur 3 Bits ab 30
  uint16_t year  = ((cp48 >> 29) & **0x07**) + (((cp48 >> 36) & 0x0F) * 8) + 2000;       // bits 30, then 32-37 to 40

Anbei dir Firmware.bin und index.html zum Test.

[Firmware1-4-4.zip](https://github.com/user-attachments/files/19997243/Firmware1-4-4.zip)

Poldi8488, 2025-05-01:
Ich hab die Test-FW mal aufgespielt und im IO-Broker Datenpunkt (Amis_Time) und Skript angepaßt.
Soweit zählt der Datenpunkt im Sekundentakt hoch. Damit kann ich jedenfalls meine Überwachung des Datenempfangs bewerkstelligen.

Gegenüber der "Unix-Epoch-Time", die ich im IO-Broker von der Systemzeit generiere, differiert der Counter allerdings, wobei auch noch zu berücksichtigen ist, daß die Epoch-Time in ms zählt.

Hier mein modifiziert IO-Broker-Script:
const idJson = 'mqtt.0.AMIS'; 
const id180 = '0_userdata.0.AMIS.180'; 
const id280 = '0_userdata.0.AMIS.280'; 
const id381 = '0_userdata.0.AMIS.381'; 
const id481 = '0_userdata.0.AMIS.481'; 
const id170 = '0_userdata.0.AMIS.170'; 
const id270 = '0_userdata.0.AMIS.270'; 
const id370 = '0_userdata.0.AMIS.370'; 
const id470 = '0_userdata.0.AMIS.470'; 
const id11280 = '0_userdata.0.AMIS.11280'; 
const idsaldo = '0_userdata.0.AMIS.saldo';
const idAmis_Time = '0_userdata.0.AMIS.AMIS_Time'
on(idJson, function(dp) { 
 let obj = JSON.parse(dp.state.val); 
 setState(id180, obj['1.8.0']/1000, true); 
 setState(id280, obj['2.8.0']/1, true); 
 setState(id381, obj['3.8.1']/1000, true); 
 setState(id481, obj['4.8.1']/1000, true); 
 setState(id170, obj['1.7.0']/1, true); 
 setState(id270, obj['2.7.0']/1, true); 
 setState(id370, obj['3.7.0']/1, true); 
 setState(id470, obj['4.7.0']/1, true); 
 setState(id11280, obj['1.128.0'], true);
 setState(idsaldo, obj['saldo'], true);
 setState(idAmis_Time, obj['time'], true);
}); 

[image]

mgerhard74, 2025-05-02:
Version 1.4.4 erstellt. lg Gerhard

closed 2025-05-02 by mgerhard74