mgerhard74/amis_smartmeter_reader #17: MQTT Verbindung nicht nur zu lokalen IPs sondern auch zu FQDN

mgerhard74, 2025-05-15:
Hallo Zusammen, 
ich brauche wieder mal eure Hilfe. Es gab schon öfters den Wunsch, per MQTT auch an FQDN-Mqtt Server Adressen zu verbinden.

Ich habe die letzten Zeilen in mqtt.cpp geändert, funktioniert nur leider immer nur mit IP-Adressen, nicht mit FQDN:
```

void mqtt_init() {
  File configFile = LittleFS.open("/config_mqtt", "r");
  if(!configFile) {
    DBGOUT(F("[ WARN ] Failed to open config_mqtt\n"));
    writeEvent("ERROR", "mqtt", "MQTT config fail", "");
    return;
  }
  DynamicJsonBuffer jsonBuffer;
  JsonObject &json = jsonBuffer.parseObject(configFile);
  configFile.close();
  if(!json.success()) {
    DBGOUT(F("[ WARN ] Failed to parse config_mqtt\n"));
    writeEvent("ERROR", "mqtt", "MQTT config error", "");
    return;
  }
  //json.prettyPrintTo(Serial);
  if (json[F("mqtt_enabled")]) {
//    mqttClient.onSubscribe(onMqttSubscribe);
//    mqttClient.onUnsubscribe(onMqttUnsubscribe);
    mqttClient.onMessage(onMqttMessage);
    mqttClient.onConnect(onMqttConnect);
    mqttClient.onDisconnect(onMqttDisconnect);
    mqttClient.onPublish(onMqttPublish);
    String broker = json["mqtt_broker"].as<String>();
    int port = json["mqtt_port"].as<int>();
    IPAddress ip;
    if (ip.fromString(broker)) {
      mqttClient.setServer(ip, port);
      } else {
      mqttClient.setServer(broker.c_str(), port);
      }
    mqttTimer.once_scheduled(15, connectToMqtt);
  }
}
```

lg Gerhard

Kriechi, 2025-05-18:
Ich vermute du musst den `String broker` als reference behalten. Aktuell wird der `c_str` am Ende der `mqtt_init` function aufgeräumt und die MQTT client library verliert den eigentlichen Inhalt des broker server names den du mit `setServer` gesetzt hast. Du könntest z.b. mit `malloc` einen Speicherbereich holen, den broker string rein kopieren, und dann überlebt er auch das Ende der `mqtt_init` function.

mgerhard74, 2025-05-18:
Danke, ich hätte folgendes probiert:
```

char* mqtt_broker_address = nullptr;
void mqtt_init() {
  File configFile = LittleFS.open("/config_mqtt", "r");
  if(!configFile) {
    DBGOUT(F("[ WARN ] Failed to open config_mqtt\n"));
    writeEvent("ERROR", "mqtt", "MQTT config fail", "");
    return;
  }
  DynamicJsonBuffer jsonBuffer;
  JsonObject &json = jsonBuffer.parseObject(configFile);
  configFile.close();
  if(!json.success()) {
    DBGOUT(F("[ WARN ] Failed to parse config_mqtt\n"));
    writeEvent("ERROR", "mqtt", "MQTT config error", "");
    return;
  }
  if (json[F("mqtt_enabled")]) {
    mqttClient.onMessage(onMqttMessage);
    mqttClient.onConnect(onMqttConnect);
    mqttClient.onDisconnect(onMqttDisconnect);
    mqttClient.onPublish(onMqttPublish);
    String broker = json["mqtt_broker"].as<String>();
    int port = json["mqtt_port"].as<int>();
    IPAddress ip;
    if (ip.fromString(broker)) {
      mqttClient.setServer(ip, port);
    } else {
      if (broker != nullptr && broker.length() > 0) {
        delay(2000);
        //IPAddress resolvedIP;
        //if (WiFi.hostByName(broker.c_str(), resolvedIP)) {
        //  if (config.log_sys) writeEvent("INFO", "mqtt", "DNS check. Broker " + broker + " IP is " + resolvedIP.toString(), "");
        //} else {
        //  if (config.log_sys) writeEvent("WARN", "mqtt", "IP of " + broker + " could not resolved ", "");
        //}
        // Speicher bereinigen, falls vorher schon belegt
        if (mqtt_broker_address) {
          free(mqtt_broker_address);
          mqtt_broker_address = nullptr;
        }
        // Speicher für broker-Adresse reservieren und kopieren
        mqtt_broker_address = (char*) malloc(broker.length() + 1); // +1 für Nullterminierung
        if (mqtt_broker_address != nullptr) {
          strcpy(mqtt_broker_address, broker.c_str());
          mqttClient.setServer(mqtt_broker_address, port);
        } else {
          DBGOUT(F("[ ERROR ] malloc failed for mqtt_broker_address\n"));
        }
      }
    }
    mqttTimer.once_scheduled(15, connectToMqtt);
  }
} 
```

Mit einer IP Adresse als MQTT Broker gibt es kein Problem, wenn allerdings ein Hostname verwendet wird, hängt der Lesekopf in einer Reboot Schleife....

Kriechi, 2025-05-18:
In welcher Form gibst du den Hostname an? Mit scheme davor? mit `:1883` dahinter? Oder wirklich als reiner FQDN mit/ohne trailing dot?

DNS funktioniert - du hast ja schon debug code dafür?

mgerhard74, 2025-05-18:
Den MQTT Broker habe ich mit "Rechnernamen" angegeben. Leider konnte ich noch nicht ausgeben, ob die DNS Auflösung funktioniert, da der Lesekopf dann in einer Reboot-Schleife hängt (Immer wenn MQTT Broker keine IP Adresse ist).

mgerhard74, 2026-09-15:
Ich habe die Info bekommen, dass dies nun in der aktuellen FW funktioniert.

closed 2026-09-15 by mgerhard74